EdgeTX/edgetx-sdcard #8: /SCRIPTS/TOOLS/ content

JimB40, 2021-06-15:
I think we should discuss & agree what we include in terms of *.lua tools in 2.4.0 Release Candidate
crossfire.lua is old version of TBS configuring tool as an example

pfeerick, 2021-06-16:
Especially when it may be better for people do download it if they actually want it, and in doing so, get the current version. i.e. I don't use either, so why do I want the scripts for them included? 

Maybe it could be split also - lite/normal and maxi pack... one being like just the stock widgets, sounds, etc, and then the other is more is like LUA and widget overload ;)

JimB40, 2021-06-16:
Changing my mind :) not much time in terms of Release Candidate deployment date.
SD card content should be just reviewied at this stage.
1. Files not needed - deleted (in /IMAGES, /WIDGETS etc)
2. Stuctucture retained to,preserve backward compatibility
3. Folders and files within new structure added (if anything prepaared by @lshems)
4. Everything tested aginst possible file-missing bugs

Who's addressing this?

lshems, 2021-06-16:
I'm running into some time issues to contribute.there is no hurry either, as the current opentx way is working. Just not optimal.

CoderElectronics, 2021-06-24:
@JimB40 I will fix all those changes in a Commit today

CoderElectronics, 2021-06-24:
@JimB40 

for #1 what images can be deleted?
for #2 what are we trying to maintain backwards compatibility with? OpenTX 2.3?

JimB40, 2021-06-24:
#1 All of them. They are random pictures of old planes and quads. If time will allow we can search for 1 genenric picture for category representative ie Plane/Glider/Quad/Car/Boat etc. or put gfx with icon of it.

offer-shmuely, 2021-12-02:
I think it is easier for the common user to have it all in a single place and delete/disable the once he does not want (can we a way to disable widget without delete them?)
instead of search and collect widget from different sources?
for the advance user it does not matter anyway, I can collect mine, write them, or delete them, it soes not matter.
but many times I need to come with the computer to the field, and give a service of updating the widgets/sounds/images for the more technofob fliers

pfeerick, 2022-05-10:
Closing as images were cleared out a fair while ago, and only "current" Lua is currently being distributed. More Lua tools and widgets may be added in the future, but not as yet.

closed 2022-05-10 by pfeerick