EdgeTX/edgetx-sdcard #12: not possible to unzip on ubuntu??

lshems, 2021-06-19:
Lshems@Khalil-Surface:~/edgetx$ wget https://github.com/EdgeTX/edgetx-sdcard/releases/tag/latest/horus.zip 
--2021-06-19 22:25:03--  https://github.com/EdgeTX/edgetx-sdcard/releases/tag/latest/horus.zip
Resolving github.com (github.com)... 140.82.121.3
Connecting to github.com (github.com)|140.82.121.3|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Saving to: ‘horus.zip’

horus.zip               [  <=>               ]  94.96K   359KB/s    in 0.3s    

2021-06-19 22:25:04 (359 KB/s) - ‘horus.zip’ saved [97234]

lshems@Khalil-Surface:~/edgetx$ unzip horus.zip
Archive:  horus.zip
  End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.
unzip:  cannot find zipfile directory in one of horus.zip or
        horus.zip.zip, and cannot find horus.zip.ZIP, period.
lshems@Khalil-Surface:~/edgetx$

CoderElectronics, 2021-06-24:
I'm not sure why this is happening, I tried to replicate it on Ubuntu 18.04 and it doesn't show. The Actions workflow is using normal Ubuntu ZIP to create the archives so they are to my knowledge all standard. Does this occur with any other zip files?

pfeerick, 2021-06-25:
@lshems You used the wrong URL to download... the size of the zip file is your first clue... its gone from a multi-megabyte sized file to a 94.96K sized file?!? Also, if you open it with any text editor, you'll see it is a HTML file. You downloaded the web page for the release, not the zip file. ;)

Another trick you can use is the `file` command, which will try to identify what type of file you have... i.e. for the file you downloaded
```
$ file horus.zip
horus.zip: HTML document, UTF-8 Unicode text, with very long lines
```

For the correct file:

```
$ file horus.zip.1
horus.zip.1: Zip archive data, at least v1.0 to extract
```

Just some hints to help you on the inevitable next time something doesn't work as expected. 

After checking the actual zip file links on the releases page, you really want:
`wget https://github.com/EdgeTX/edgetx-sdcard/releases/download/latest/horus.zip`

btw, to prevent strikethrough and other weirdness on logs and such, use the triple backtick (```) markdown formatting command before and after your log / block of code to prevent any formatting codes been triggered.

lshems, 2021-06-25:
Thanks! Closing

closed 2021-06-25 by lshems