EdgeTX/edgetx-sdcard #14: Open TX Models Not Recognised

limitingKano, 2021-06-24:
Hi,

Just to let you know I have been unable to get my previous OpenTX models recognised and converted.

Radio Radiomaster TX16s

Action Taken: Followed install instructions onto clean, formatted SD card, then copied over models.
Result           : Model folder appears populated with Model Bin files but no models are able to be accessed no conversion message shown.

Action Taken: Install onto SD card with everything apart from OpenTX models removed.
Result           : Model folder appears populated with Model Bin files but no models are able to be accessed no conversion message shown.

Apologies if this is not a bug and an error on my side.

CoderElectronics, 2021-06-24:
I'm going to move this over to the actual Firmware repo, it's most likely not on the SD Card side.

CoderElectronics, 2021-06-24:
@raphaelcoeffic Can you transfer this issue?

closed 2021-06-24 by CoderElectronics

limitingKano, 2021-06-24:
will do