EdgeTX/edgetx-sdcard #26: Wizards: Delta Wing Wizard

pfeerick, 2021-09-29:
There was a request for a delta wing wizard

offer-shmuely, 2023-04-25:
I do not have a wing, but still like to try and do that.
What is needed for a wing?

- Motor - yes/no
- Elevron - always
- Dual rate -yes/no
- safety switch on SF (if motor exist) 

What more?

pfeerick, 2023-04-26:
The primary difference will be be the elevons, and no aileron/elevator/flaps, so yeah, pretty much what you listed. I find it interesting the model wizard in Companion asks if there is a rudder. It's not common, but It does seem to be a thing.

offer-shmuely, 2023-04-26:
PR #107

offer-shmuely, 2023-04-26:
[image]

[image]
[image]
[image]
[image]
[image]
[image]

offer-shmuely, 2023-04-27:
Is it worth to add a model name as well to the wizard?

offer-shmuely, 2023-08-13:
will be closed when #107 will be merged

PR #107 (2023-09-02): wing wizard & plane wizard & dirs
edit: this PR started as wing only, but now it contain also enhancement to the plane wizard & directory restructure

Resolves #26 

adding a wizard for wing (per old request)
also added a touch support for next/previous pages
maybe need better pictures (currently used the plane ones)

[image]

[image]
[image]
[image]
[image]
[image]
[image]
closed 2023-09-02 by pfeerick