EdgeTX/edgetx-sdcard #36: BattCheck (Analog) is not saved on model

TheIsotopes, 2021-12-16:
@offer-shmuely I tested Battcheck on opentx and edgetx, on both the widget is not saved in the model.
When i change the model and go back, the widget is no longer displayed.

pfeerick, 2021-12-16:
Ok, it's a name issue... radio is only saving `"BattCheck (A"` instead of `"BattCheck (Analog)"`. If you change 

https://github.com/EdgeTX/edgetx-sdcard/blob/3e27d753fb4b7e584d8d10f982b77ae7783aa801/sdcard/horus/WIDGETS/BattCheckAn/main.lua#L575

to `name = "BattCheckAn"` and try again it will be fine. I'll have a look at the code either tomorrow or on the weekend and see if that can be extended going forwards.

TheIsotopes, 2021-12-16:
thx ... working now 👍

offer-shmuely, 2021-12-18:
ooops
So the name size is enforced on saving (logical) but not on the reading from SD
I missed that one. sorry

Changing to 10 char name
lets use BattAnalog, it look the best on 10char name limitation.

TheIsotopes, 2021-12-18:
@offer-shmuely The name looks good and everything works as it should.

closed 2021-12-19 by pfeerick