EdgeTX/edgetx-sdcard #40: Lua script for Open Location (Google+) Code

ChrisOhara57, 2021-12-27:
To Peter Feerick

sorry I didn't succeed in putting a pull request.
Here is the Zip file.

I tested the code on my Taranis x9lite S.
I think it should run as a telemetry script on all black and white screens.

Thanks Chris

[gplusl.zip](https://github.com/EdgeTX/edgetx-sdcard/files/7781723/gplusl.zip)

pfeerick, 2022-01-17:
I haven't forgotten this, and will add it in the next couple of days. I just wanted to draw to your attention that it looks like there is also a widget for colorlcd radios as well ;) https://github.com/kristjanbjarni/opentx-widgets

ChrisOhara57, 2022-01-17:
Thanks for the info (at the moment I only use a x9lite) but it's good to know.
If you think there is no need for a second script, feel free to dismiss it.

pfeerick, 2022-01-23:
I just double checked it on X9D and TPro simulator, and it seems to work fine on ETX also. I've added a commit adding this with you as the commit author.

commit 5fe907a5 (2022-01-23): [b&w] Add GPS Plus Code telemetry widget
Resolves #40
closed 2022-01-23 by pfeerick