EdgeTX/edgetx-sdcard #46: BattCheck (analog) widgets disappear on model switch and reboot

wucke13, 2022-01-26:
**Describe the bug**
If I add the BattCheck (analog) widget to my telemetry screen, it disappears once I reboot the remote or switch to another model and back. 

**To Reproduce**
add the BattCheck (analog) widget to a telemertr screen of a model, switch to another model, switch to the previous model

**Expected behavior**
the widgets stay where I configured them

**General information**
 - EdgeTX 2.6.0 (338d4d2c)
 - FrSky X10S

pfeerick, 2022-01-26:
How is the name coming up - As it sounds like you may be running an old version of the widget = > https://github.com/EdgeTX/edgetx-sdcard/issues/36

wucke13, 2022-01-27:
Yes I can confirm this is a duplicate of #36 . I just unzipped the new version of the sdcard over the existing files with overwrite. However, this left me with three widgets,

+ `BattAnalog`
+ `BattCheck`
+ `BattCheckAn`

And unfortunately the old one took precedence, I guess.

closed 2022-01-27 by wucke13