EdgeTX/edgetx-sdcard #58: Grupner HoTT.lua missing in SCRIPTS/TOOLS

mha1, 2022-03-24:
Hi,

would be nice to add "Graupner HoTT.lua" to SCRIPTS/TOOLS for MPM capable radios like the TX16s (Horus.zip). Please find the OpenTX 2.3.14 version in attached zip [Graupner HoTT.zip](https://github.com/EdgeTX/edgetx-sdcard/files/8344236/Graupner.HoTT.zip).

Thanks,
Michael

pfeerick, 2022-03-25:
I see your point, but I would instead suggest it is better to get the _current_ version of the script, rather than added it to the sd cards here, and then have to keep updating it. It is actually part of the MPM lua scripts package... => https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/tree/master/Lua_scripts , and in the near future Flasher (and hopefully buddy) will allow you to update existing scripts (Flasher already lets you install the current version of the MPM, ELRS, and TBS Agent lite scripts).

mha1, 2022-03-25:
I guess my point is to make the process of first install or updating the radio as convenient as possible. A one point tool like flasher is the right way to do this. But why not do both. Having a fully set up SD Card doesn't harm flasher, does it?

Btw: I wasn't happy to start this as issue, wanted it more like a discussion. And it is - as you say - more about the philosophy of to do things. Isn't there a better way in git or elsewhere to separate issues from proposals? Issues should be for hard error reports

closed 2022-10-01 by mha1