EdgeTX/edgetx-sdcard #62: New plane/drone/glider wizards do not run

Luznatural, 2022-05-08:
I have a Taranis X9D+ SE 2019, and a Radiomaster TX16S, both with EdgeTX 2.7 an the latest SDcard 2.7 for Taranis X9, and Horus.
I have checked that the correspondant .lua files for planes/drones/gliders/ are in the SCRIPTS/WIZARD folder on the SD. The bmp files for the wizards are also there.
When you hold enter, select create a new plane/drone/glider in an empty memory slot, the wizard to help you does not start at all. Directly gives you a generic MODEL name and remains there.
You have to manually select the configuration you want.
Please, If I am doing something wrong I would appreciate any help.
Thanks in advance

pfeerick, 2022-05-09:
You need to launch the wizard manually... it intentionally no longer runs automatically by design, as more people were frustrated with the wizard being there than those who wanted it. 

I'll double check the wizard loader which is used for the smaller screen radios tonight, and if that works with the X9 radios without issue, update your SD card tomorrow, and you'll have a "Wizard Loader" option in the SYS -> Tools menu, which you run after creating a new model if you want to use a wizard.

pfeerick, 2022-05-09:
https://github.com/EdgeTX/edgetx-sdcard/commit/4e951cdc0e19172977bf31de22aee2bc98b77699 adds the loader script used on the 128x64 B&W radios to the X9 radios also.

closed 2022-05-09 by pfeerick

Luznatural, 2022-05-09:
Thanks for your help.
It worked perfectly in the way you described it