EdgeTX/edgetx-sdcard #68: Include all standard folders

elecpower, 2022-07-28:
The ETX issued sd card images should be complete to avoid other issues especially if folders are renamed, dropped, etc

pfeerick, 2022-07-29:
Are there any specific ones you have in mind? i.e. RADIO and MODELS aren't included, since the radio will create those itself, but there is isn't any specific reason they couldn't be added other than preventing collision with pre-existing folders. Other than that, between the GLOBAL and the radio family specific folders I think they're all there? *(Hm... ok, NV14 doesn't have TEMPLATES folder...)* ... 😮

elecpower, 2022-07-29:
Yes the radio creates MODELS and RADIO when it boots and loads ETX. However.
Companion searches for RADIO folder as a top level folder in the SD card drives connected to the computer. So if you have downloaded the card image, copied to sd card, plugged card into radio and performed a 3 finger boot the sd card does not yet have the RADIO and MODEL folders and thus Companion reports it cannot find the radio.
So I'd prefer to have them as empty folders on all images.
Also some users may want to preload the sd card before inserting into the radio for the first time and having the folders already created avoids some user typos and hidden spaces etc.

PR #71 (2022-08-11): feat: Add default radio folders and READMEs
Resolves #68
closed 2022-08-11 by pfeerick