EdgeTX/edgetx-sdcard #69: S2 not working properly on RM T16s MK II in v2.7.1

olli-e, 2022-08-04:
When installing 2.7.1 on my RM T16S Mk II, S2 is only using half the way - it can either be -100 to 0 or 0 to +100, but not the full range as S1 does. I calibrated and did everything possible after blank install and re-tried three times, but it is always the same - S2 is not fully usable. Fell back to v2.7.0 and everything is working again. Just as an Info - will be using v2.7.0 until fix.

Thanks for all your hard work fellows!

pfeerick, 2022-08-04:
Hm... it sounds like there could be something more specific to your radio going on... I have both a pre-production and a final release TX16SMKII, and neither have any issues with the S1 or S2 pot movement... i.e. if I go to the Radio Settings -> Hardware > Analogs page (the button is right at the bottom), 05 (S1) and 07 (S2) move the full range from -100 - 0 - 100, (0-2048 calibrated, 0-4096 raw +/- 2-3 count). 

When you did the calibration, did you centre the pots, and then do the full range movement? If you didn't centre them first, that would cause this. You must centre *all* analog inputs when prompted (on the TX16S, that is the sticks, pots, sliders and the 6POS switch at either 3 or 4), and then move them all full range.

olli-e, 2022-08-05:
Very strange - yes, I centered them before doing calibration - as I wrote, I repeated that step over and over, just to exclude any possible mistake on my side - you, how this is, usually the error is in front of the PC/Tx ;-) However, to exclude I have a hardware error, I installed OpenTX - works; then 2.7.1, didn't work - then downgrade to 2.7.0 - works again. Tbh, doesn't sound to me like there's something strange thing with my Tx going on, but I'm no expert, just from the logics. BTW, it's a brand new MKII.

pfeerick, 2022-08-05:
Yes, I've certainly been the cause of some errors 😆. Just want to make doubly sure it wasn't something simple ;) 

I'll have a check of the code, but I can't off the top of my head think of anything that would have changed in 2.7.1 that would cause it... but the consistency of downgrading (as well as trying OTX) to fix it certainly suggests something is wrong there - although given I can't reproduce it that makes it even harder to pin down. 

Can you upload a zip of your RADIO and MODELS folder so I can see if applying that to mine causes it to play up.

olli-e, 2022-08-05:
You mean, I should once again flash 2.7.1 and see if that problem still exists and then upload those files? BTW, don't know if that is also interesting - the 2.7.1 flasher never identified any SD Card on my PC, also it's visible in the file explorer and after I flashed with the edge flasher I always got Bad Data and needed to manually flash using the bootloader - even with a blank SD Card. Sorry, if I mix two thing sup now.

pfeerick, 2022-08-05:
lol... I just realised this is in the sdcard repo ... moving it to the main edgetx repo. 

So, basically yes, jump back to 2.7.1, and make a copy of those folders if the problem is still there. 

For the SD card issue, was this with the stock card? As you may find with other cards you need to format the card so that its not in in a weird exfat format... you want plain FAT16/FAT32, with 4K sectors if you get a choice.