EdgeTX/edgetx-sdcard #73: Add LI-ION battery voltage table in BATT ANALOG Widget

DG9BFC, 2022-08-24:


offer-shmuely, 2022-08-24:
can we agree the 3.0v is 0%, and 4.1 is 100%
and we can take from the graph the other values
we can also see what the battery checker of the ISDT is presenting

DG9BFC, 2022-08-24:
3v as empty leaves enough safety margin ... so yes i think thats ok
we could go down lower (2.8) but for long cell life its better to stay a bit higher and that rest charge from 3 to 2.8 is not that much
The curve is dropping a bit faster at top end (4.1 to 4 or 3.9 ... call it first quarter) ... then slower over the next say 50% (down to around 3.5) .. and then faster and faster till it falls down at 3v
no need to be super precise 
as you both know it also depends on the cell type, age of the cell, temp, load and a lot of other things (as an example i use 2x21700 in parallel (9 to 10Ah) a
and so the load on the transmitter battery is below 0.05c
if you use a single 18650 (two in a row) we have a load of maybe 0.2c
so voltage measured will be higher on bigger packs (lower drop off of the voltage caused by the lower internal resistance)
3v is 0% with some rest load for safety and cell life
3.5v should be around 25 to 30%
3.75 is around 50 to 60%
4.1 is fully charged
something like that ... 
just look "a few" of those curves and find a compromize

DG9BFC, 2022-08-24:
sidenote;
would it be possible to show two digits behind the comma??
as you know the voltage drop is low in middle of the charge ... 3.5 is much lower as 3.59
and showing two digits would better represent whats going on in your battery pack
if space is tight then show that second digit only in main screen  and fullscreen view but not in the top row (just an idea from a dumb  nut hi hi)
i can only see that second digit in the battery calibration box ... showing that also in main view would be superb

offer-shmuely, 2022-08-26:
I used it once, did not find it useful (for me)
Do you think it worth to overwhelm with more data when the it is not accurate in the first place?

DG9BFC, 2022-08-26:
percentage of charge  may not be accurate (cause it depends also on used cells, current, et  etc) ... but..... voltage is voltage so that number is accurate (or should be)
so how accurate the "SOC" is depends how well you make that curve ... but the numbers in volt ... that should be more or less exact (if you calibrated it in system settings)
so .. yes i would like to see that additional digit if there is enough space on the screen
its not overwhelming ... its just a finer resolution :-)

offer-shmuely, 2022-08-26:
added the 2 digit after dot to here (for now)
lets check it on a day to day use
https://github.com/offer-shmuely/edgetx-x10-widgets/blob/master/BattCheck/main.lua

pfeerick, 2022-08-27:
>  but the numbers in volt ... that should be more or less exact (if you calibrated it in system settings) so .. yes i would like to see that additional digit if there is enough space on the screen its not overwhelming ... its just a finer resolution :-)

For the Batt(Cell)Check widget, 2 digits after the decimal sounds perfectly fine, since these are the individual readings for each of the cells as measured by something like the Frsky FLVSS. If there is much point to that given SoC is supposed to be the main metric is another question. 

But for the BattAnalog widget (which is the focus here, isn't it?) it allows for showing of total voltage (at which point two digits after the decimal would be as accurate as the measurement accuracy allows) but also allows for approximated cell voltage, which is *not accurate at all* since it relies on all the cells being perfectly equal... so I would argue that when showing approximate cell voltage, no more than one digit after the decimal should be shown due to inaccuracy.

DG9BFC, 2022-08-27:
batt analog is used for (as an example) tx battery (two identical cells in a row) so the voltage of each cell should not differ tooo much (if you use identical cells) ...
i would like to see two digits behind decimal ... IN BOTH WIDGETS!!
the result may not be exact but ... what if you have (as an example) 7.9V?? what to show then?? 3,4?? (surpress the second digit) 3.5?? (rounding up last digit)
in the above numbers you see how much it differs if you surpress a digit or round up/down a digit ... so yes i would say we need two digits behind decimal
even if the result may not be perfect it is better as without that number

DG9BFC, 2022-08-27:
if space is tight ... let away the "V" cause any user should know that a "V"oltage is shown in volt not apples, oranges or bananas hi hi

DG9BFC, 2022-08-31:
i tried the 4 days old version with li-ion selection but there seems a fault
lipo works
li-ion not

offer-shmuely, 2022-08-31:
Can you give more info?
What was the widget settings?

offer-shmuely, 2022-09-01:
can you send me the values that helped you create your chart?

DG9BFC, 2022-09-01:
maybe needs some finetune?!?
------------------------------------------------------------------------------------

-- from: DG9BFC Sigi (dg9bfc@freenet.de)
local _liionPercentListSplit = {
  { {2.80,   0 },{2.85,   1 },{2.89,   2 },{2.92,   3 },{2.94,   4 } },
  { {2.96,   5 },{2.97,   6 },{2.98,   7 },{2.99,   8 },{3.00,   9 } },
  { {3.01,  10 },{3.02,  11 },{3.03,  12 },{3.04,  13 },{3.05,  14 } },
  { {3.06,  15 },{3.07,  16 },{3.08,  17 },{3.09,  18 },{3.10,  19 } },
  { {3.11,  20 },{3.12,  21 },{3.13,  22 },{3.14,  23 },{3.15,  24 } },
  { {3.16,  25 },{3.17,  26 },{3.18,  27 },{3.19,  28 },{3.20,  29 } },
  { {3.21,  30 },{3.22,  31 },{3.23,  32 },{3.24,  33 },{3.25,  34 } },
  { {3.26,  35 },{3.27,  36 },{3.28,  37 },{3.29,  38 },{3.30,  39 } },
  { {3.31,  40 },{3.32,  41 },{3.33,  42 },{3.34,  43 },{3.35,  44 } },
  { {3.36,  45 },{3.37,  46 },{3.38,  47 },{3.39,  48 },{3.40,  49 } },
  { {3.41,  50 },{3.42,  51 },{3.43,  52 },{3.44,  53 },{3.45,  54 } },
  { {3.46,  55 },{3.47,  56 },{3.48,  57 },{3.49,  58 },{3.50,  59 } },
  { {3.51,  60 },{3.52,  61 },{3.53,  62 },{3.54,  63 },{3.55,  64 } },
  { {3.56,  65 },{3.57,  66 },{3.58,  67 },{3.59,  68 },{3.60,  69 } },
  { {3.61,  70 },{3.62,  71 },{3.63,  72 },{3.64,  73 },{3.65,  74 } },
  { {3.66,  75 },{3.67,  76 },{3.68,  77 },{3.69,  78 },{3.70,  79 } },
  { {3.71,  80 },{3.72,  81 },{3.73,  82 },{3.74,  83 },{3.75,  84 } },
  { {3.76,  85 },{3.77,  86 },{3.78,  87 },{3.79,  88 },{3.80,  89 } },
  { {3.81,  90 },{3.83,  91 },{3.83,  92 },{3.84,  93 },{3.85,  94 } },
  { {3.87,  95 },{3.89,  96 },{3.92,  97 },{3.96,  98 },{4.04,  99 } },
  { {4.15, 100 }
}

-----------------------------------------------------------------------

this curve is for a bit higher current so i guess it should be finetuned (in real stays higher in voltage for a longer time and then drops even faster at the end)
chopped off at 2.8v (3v is 9% so we have "alarm" at below 10% charge)
alarm on at 3.1 would be "below 20%" ... and so on
................................................
anyway ... all that can only work if you MEASURE 2 digits and after divide by cell number look in the above table ... with only using a single digit it is way way to coarse stepping
[image]

DG9BFC, 2022-09-01:
at best use THREE digits after divide by cell number ... see the other table a bit higher in the widget (lipopercentlistsplit)... there are three digits behind decimal ... not?!?
if it WORKS with three digits (internal ... no need to display all three on screen) ... THEN we can finetune the curve depending on the cell used (or find a working compromize that fits most li-ion round cells) ...
cause the voltage drop (especially on low loads like in transmitter) IS VERLY LOW over the state of charge you MUST use as many digits as you can get from the sensor ... and if divide by cell number gives you a third digit ... do not throw away that info
only then we can have fine steps in the percent values
a single digit behind decimal is to coarse

DG9BFC, 2022-09-01:
is my table "tooo big" ???
i see you chopped away some values in your widget (iirr above 90% ?!?!?)

DG9BFC, 2022-09-01:
above you see my email addy ... maybe we should clear this in direct mail contact?!?!?

offer-shmuely, 2022-09-01:
the value internally is not kept in 1 or digit precession, this is only a presentation issue.
we still need to work to the percent conversion table, but for now I put any voltage above 4,1 as full battery, since of the Li0Ion battery can be charge to 4.1v per cell, and some can be charge to 4.2v
i.e.

4.10 == 100%
4.15 == 100%
4.18 == 100%
4.20 == 100%

DG9BFC, 2022-09-01:
aahhh i found a typo in my table ;-)

DG9BFC, 2022-09-01:
[image]

.....
edited by hand in the lua file on my end ... 
a value tha tis double may cause a fault :-)

DG9BFC, 2022-09-01:
sorry for that typo (but that shows that you just used drag and drop hi hi)

DG9BFC, 2022-09-02:
after editing both widgets a bit its now working better
(yes also your other widget changet a bit here)

DG9BFC, 2022-09-02:
top end of table modded or it causes an error when booting while charging
  {3.81, 90 },{3.82, 91 },{3.83, 92 },{3.84, 93 },{3.85, 94 },
  {3.90, 95 },{3.95, 96 },{4.00, 97 },{4.10, 98 },{4.15, 99 },
  {4.20, 100}
  }

DG9BFC, 2022-09-02:
values above 8.2 (after divide by cell 4.1) caused a crash
with above numbers its fine

DG9BFC, 2022-09-02:
[battwidgets.zip](https://github.com/EdgeTX/edgetx-sdcard/files/9474945/battwidgets.zip)
this is what i use now
no idea if i made a fault somewhere but it seems to work
greetz sigi

DG9BFC, 2022-09-02:
i have thrown out the old table (not needed here)

DG9BFC, 2022-09-02:
[image]

maybe now clear what i mean with use not only a single digit??

DG9BFC, 2022-09-02:
tiny fault -- should be 43 :-)

offer-shmuely, 2022-09-02:
it is not the widget that reduce the resolution, it is the c++ api of the main firmware
you need to ask for a change there

this function:
local v = getValue(wgt.options.Sensor)

you can see this log:
log(wgt.options.source_name .. ", value: " .. fieldinfo.name .. "=" .. v)

offer-shmuely, 2022-09-02:
I updated the widgets with the value you suggested for li-ion, except that I keep the 4.1v as 100%

I am not changing the values of the lipo, since we already had discussion on them in the past, and they where tested and found to be ok

DG9BFC, 2022-09-02:
ok about the li-ion values (mine where probably a bit too optimistic)
4.1v as 100% gives and error if vóltage is higher (but maybe you fixed that now??
at least in the other version it gave an error when connected to charger while booting
and even after disconnect for a while you had to use it a bit to lower voltage and then do a reboot for the error to be away
thats why i made the top end change (4.2=100%)

DG9BFC, 2022-09-02:
... about the coarse stepping ... where do i have actually to ask??
write an issue in the main edge tx fw??
ok i can do that (but i ask you before doing so)

DG9BFC, 2022-09-02:
tested your version .. shows an error
ERROR in refresh();?:0: attempt to compare number with tabl
i see you again use "double brackets" (my version did not have them)
...
or what else can cause that error (i do know nothing about "lua grammar and signs like brackets and commas etc etc)

greetz sigi dg9bfc

DG9BFC, 2022-09-03:
side note ... who said that my li ion table is correct?? ...
there is a reason why i edited the lipo table (that was made BEFORE you added the second table)
i kept it to see what works best (call it experimental mode)
and it looks like the other (earlier) table matches better ... especially cause we see only one digit from the main fw...as an example see picture
after a bit of use the voltage dropped to 7.9 ... divide by 2 is 3.95 but in the widget it is shown 4.0 (rounded up to next value) ... so it shows 97% but in real it is maybe down to 80??

[image]

ame thing on longer use ... dropped to 7.7 but widget shows 3.9 as single voltage (3.85 rounded up) and so shows 95% (but should be 70 or so)
 ... 
so i am still testing out the numbers ...

DG9BFC, 2022-09-03:
on left you see the older table on the right you see the new table (shows tooo high state of charge i would say)
...
it says 7.7v as long as it is above 7.6v ... (also when i play the audio to tell me the battery voltage) ... so it is rounding upwards (also 7.61 is shown as 7.7)

offer-shmuely, 2022-09-03:
Looks like you do the test with RxBatt
Try to do it with A1/A2
I think that rxbat have some logic behind it.

i also that  you still see only one digit behind the dot, probably i did not replaced on all places, i will fix it.

i chanced the  values to groups (what you call double brackets) on all the 4 places, this method is more complicated, but faster run time (less load on cpu)

offer-shmuely, 2022-09-03:
can you send screenshot of the following:
* telemetry sensor setting
* widget setting
* how the widget look like

DG9BFC, 2022-09-03:
i do not have any telemetry (besides what the receiver gives).. using the tx battery for testing
i can connect a receiver and use that sensor but that does not look different then the tx battery view
thanks for explaining the double brackets (groups) ... another lesson learned
would you be so kind and drop me a direct mail?!? (dg9bfc at freenet dot de) .. its much easier if we do those things in direct contact ... 
if you make a change you direct can send me the main.lua file ... now i always have to copy it manually .. then upload to transmitter and test...
direct mail would be easier and go faster and when we call it "finished" you drop it here into github for the masses
also easier not to mess up github with screenshots, pictures etc (just takes space) ...
i now have to "look" when you have upload a new version (user logged in xy hours /days ago) ... 
a mail "hey i have a new version .. may you try that" ... would solve that

DG9BFC, 2022-09-03:
[image]

DG9BFC, 2022-09-04:
[image]
new values ...
seems to be a good compromize between the different li ion cell types 
main thing is the mid section between 3.8 down to 3.5 and in that area allmost all cells are quite linear and have between 30 and 70% load (give or take
below 3.4 (and then again at 3.2 and 3v )they drop fastwer and faster and then its time to recharge your battery anyway
maybe my curve needs some tiny finetune at the very upper and lower end but i guess its usable and matches the discharge curves of above cell types good enough
i tried to peel out the rough numbers from above curves and tried to find a compromize between them and to have a nice shaped curve that looks about the same as the curves we see in the data sheets
what do you think?!?

DG9BFC, 2022-09-04:
it also does not matter much if we cut at 3 v or at 2.8 ... that is the last 5 or 10 % load so nothing to worry about in real

offer-shmuely, 2022-09-12:
added a PR #78 that support li-ion battery voltage

DG9BFC, 2022-09-12:
i will download your last version and test it ... especially the 1/8th size widget gave (or still gives) me a headache ... i switched sliders and trimms and topbar off ... and back on ... and now the grey box again blocks the lower part of the white box (and numbers are not well centered anymore)
you wrote that you made the 1/8th mode adaptive (so i will check that on a real trx and not a sim) ... reports in a few minutes

DG9BFC, 2022-09-12:
[image]

DG9BFC, 2022-09-12:
i shaved your numbers a bit

DG9BFC, 2022-09-12:
[image]

offer-shmuely, 2022-09-12:
it looks like you are not using the latest one
this is the current layout on 1/8 boxs

[image]

DG9BFC, 2022-09-12:
if i switch off all outer things (topbar, flight mode and trimms /sliders) then the cell is like in 1/4th view ... ok on that layout
but now if i only switch on the top bar (horizontal cell) the text is a bit tooo high placed (showing a bigger cell now)

DG9BFC, 2022-09-12:
i used your main lua ... with copy and paste!

DG9BFC, 2022-09-12:
only changed one little thing (the space at the percent in 1/8th view)

DG9BFC, 2022-09-12:
it it possible that the height where the text is placed also changes if the height of the cell changes?? (so that it is centered on both sizes we still have on the horizontal cell)
somethink like>> place text at cell height/2 - textheight/2
cause text height is similar in both sizes the second part should be a fixed value

DG9BFC, 2022-09-12:
or plus text height/2 if you count from lower edge (if i remember right the coordinates start at bottom left)

DG9BFC, 2022-09-12:
like i have it finetuned now from your version looks perfect if i have fm, trim and topbar activ ... but not if i only have topbar active

DG9BFC, 2022-09-12:
or did i made it wrong with copy and paste??

DG9BFC, 2022-09-12:
[image]

offer-shmuely, 2022-09-16:
The pull request has merged
The issue can be closed

DG9BFC, 2022-09-17:
yes .. seen in on of the zip files (merged not in all versions??)
it is c480x272 but not in c480x320 ??? (checked both)
anyway ... works fine here now 
i still have a bit different numbers in screen layout but ok so be it ... i did not want the grey blinking rectangle to be drawn over the lower end of the white box (battery cell) and it seems as the simulator gives a bit different view as the real thing
about the exact numbers in voltage ... i use 4.2v cells so my upper end looks a tiny bit different
... but all that is just cosmetics (and any user can finetune the numbers to his needs)
so yes i guess we can close it here

closed 2022-09-17 by DG9BFC

pfeerick, 2022-09-17:
> it is c480x272 but not in c480x320 ??? (checked both)

Correct - that is part of another ongoing problem/discussion. And unless you have a PL18 and are running a test build of EdgeTX on it, the `c480x320` is the wrong one to be looking in anyway ;)

DG9BFC, 2022-09-17:
morning :-)
no i do not own a PL18 ... just checked if the add ons for lipo/li-ion are made in all versions