EdgeTX/edgetx-sdcard #74: S6R/S8R scripts

rdeanchurch, 2022-09-05:
There are no scripts for FrSky stabilized receivers.
Is this by design or an oversight?
They are available in OpenTx. SD contents.

pfeerick, 2022-09-06:
By design - it encourages you to ensure you have the latest version since they are supplied by the vendor - in this case Frsky.
We don't include any scripts that third parties are responsible for maintaining/supporting. This is why we don't include ExpressLRS or MPM scripts either. 

However, we are starting to add support for auto-installing the current version of some of the third-party scripts over time. i.e. Flasher will offer to auto-install MPM, TBS Agent Lite, and ExpressLRS lua when prepping the SD card. And Companion now has the option to install and keep MPM firmware and Lua up to date. So hopefully as time progresses we'll add options to also install/fetch updates for other third-party scripts on demand. But they won't be included in the repo as standard. 

The current Lua can be downloaded directly from Frsky:
https://www.frsky-rc.com/s6r/
https://www.frsky-rc.com/s8r/

offer-shmuely, 2023-01-28:
While usually I agree with pfeerick,
Since it hard to follow the changes 3rd party do to their script,
Like the fast development of elrs script, and tbs agent.

However, the S6R scripts we are talking about here, does not have the same code standard we are all used to.
FrSky does not really develop or enhance this script any more.
On some of the download links you get different versions with different changes that sometimes have different names!!
It is hard to follow what script is used for what.
It looks like some newer code of the script, is missing previous enhancements, almost like the new changes are based my mistake on old version of the code instead on top of the latest.
I do believe we need to merge all the different versions they have, and may someone will take it and enhance it, since clearly FrSky is not on the mode to improve it any more

offer-shmuely, 2023-01-28:
I will give some example...
the files you download from the S6R/S8R named FrSky SxR_R9S.lua, it is ver 2.0.1 and it good for the R6S/R8S/R9S
but the R9S, have a newer version 2.0.3 (great), but it does not under the S6S & R8S (why?)

the SR6/8/10 have versions 2.01/2.0.2/2.0.3
the version 2.0.1 here is the same 2.0.1 of the S6R, but they change the name to "FrSky STAB RX.lua" (better name!)
but the later version 2.0.3 of the S9S is not have this name change (why?)

on 2.0.2 they added a new setting, 

[image]

on version 2.0.3, they added a **new setting** (yea!!) and **remove the setting from 2.0.2** (really???)

[image]

and they also return to the  **_old name_** "**_FrSky SxR_R9S_**.lua" , is it only just me, or they forgot to do the change on top of last version????

pfeerick, 2023-01-29:
> I do believe we need to merge all the different versions they have, and may someone will take it and enhance it, since clearly FrSky is not on the mode to improve it any more

Agreed... basically, someone needs to commit to pulling together the frsky lua script mess into it's own repo and then it can either be added here or preferably at linked in lua-scripts repo and people directed to get the companion lua scripts from there in future, with the repo commit logs serving to self-explain why that is necessary.

SimagesRC, 2023-02-01:
When started going down this SxR rabbit hole, I'm glad it finally led me HERE! Although I'm really LATE to the Party regarding OpenTX because I just purchased a TX16S MkII that came with EdgeTX. For weeks I've been learning how to program this radio and will say that the radio is great however; I'm blown away with the software! For those that code all of this , I have no words except "Awesome". I know now that I've been missing out for years..... That being said; I feel rdeanchurch's pain regarding his post and the loss of some FrSky scripts for the S6R /S8R receivers. Last week I purchased 2x S6R 3-axis receivers and found very informative videos from John Hopke of "RC Video Reviews" who has mentioned you "shmuely". A bit of a learning curve because much of the videos were done with "OpenTX", but easy enough to follow along with EdgeTX. However; I had to purchase the FrSky STK because I couldn't find the SXR-calibrate.lua script to add to the tools. Atleast there's the script to make changes to the 3-axis configuration and you only have to claibrate the Rx once. Thanks pfeerick for your input and shmuely for your contributions!

offer-shmuely, 2023-07-15:
closing with PR #113

offer-shmuely, 2023-08-13:
please close as fixed
using PR #113

closed 2023-08-14 by pfeerick