EdgeTX/edgetx-sdcard #75: Extend BattCheck for FrSKY FLVS Advanced Lipo sensor (8 cells)

mha1, 2022-09-10:
@offer-shmuely: Hi Offer, FrSky has a new 8 cell capable Lipo sensor. With [PR 2302](https://github.com/EdgeTX/edgetx/pull/2302) I made a slight change to the Firmware side to have Cels deliver a max. of 8 cells instead of a max. of 6 cells. It would be very nice if you could look into extending your BattCheck LUA to display up to 8 cells. Thank you very much - Michael

PS: I did a quick and dirty change to V0.5's main.lua for my TX16s display as proof of concept but omitted the smaller displays.  This is what I basically changed (based on V0.5) to make it show up to eight cells. I am sure you can do this much nicer and also for the smaller displays.

in local function create:
```
cellDataLive = { 0, 0, 0, 0, 0, 0, 0, 0 },
cellDataHistoryLowest = { 7, 7, 7, 7, 7, 7, 7, 7 },
cellDataHistoryCellLowest = 7,
```
in local function onTelemetryResetEvent:
```
wgt.cellDataLive = { 0, 0, 0, 0, 0, 0, 0, 0 }
wgt.cellDataHistoryLowest = { 7, 7, 7, 7, 7, 7, 7, 7 }
wgt.cellDataHistoryCellLowest = 7
```
in local function refreshZoneXLarge:
```
  local pos = { { x = 101, y = 38 }, { x = 154, y = 38 }, { x = 207, y = 38 }, { x = 260, y = 38 },
	        { x = 101, y = 57 }, { x = 154, y = 57 }, { x = 207, y = 57 }, { x = 260, y = 57 } }
  ...
  local pos = { { x = 101, y = 110 }, { x = 154, y = 110 }, { x = 207, y = 110 }, { x = 260, y = 110 }, 
	        { x = 101, y = 129 }, { x = 154, y = 129 }, { x = 207, y = 129 }, { x = 260, y = 129 } }
```

offer-shmuely, 2022-09-10:
no problem, but since I could not get one (I like to...)
I need to wait until the companion simulator will support it

offer-shmuely, 2022-09-10:
actually I can simulate it in code
so here...

[image]

[image]

pfeerick, 2022-09-11:
There is a companion build for that PR, so should be able to simulate. It looks straightforward enough so is likely to be the next nightly.

mha1, 2022-09-11:
@offer-shmuely looks great, thank you very much. Where can I find the updated version?

mha1, 2022-09-11:
@offer-shmuely please try this https://github.com/EdgeTX/edgetx/suites/8233289654/artifacts/359485951 with your updated BattCheck. Thanks - Michael

offer-shmuely, 2022-09-12:
https://github.com/EdgeTX/edgetx-sdcard/pull/76

mha1, 2022-09-12:
Hi Offer,

Thank you very much. Have you tried this against with the updated Companion? Does the updated Companion telemetry simulator work for you?

And a question:
` 
cellDataHistoryLowest = {5,5,5,5,5,5,5,5},
cellDataHistoryLowestPercent = {5,5,5,5,5,5,5,5},
cellDataHistoryCellLowest = 5,
`
I interpreted this as initial indexes of cell numbers (5 for the last cell of 6 cells considered). Shouldn't the 5's be 7's now?

Thanks - Michael

offer-shmuely, 2022-09-12:
Hi
yes, I checked it with the new companion, it works great.
still need to do some alignments on the 1/4 scale widgets, but I will do it some time in the future (since now I am busy with the LogViewer that I am building)
I do think to change the mechanism of the layout to be more automatic.

when you calculating min value, you are usually math.min(new-value, old-value)
in order it to work, the initial old-value should be a high number, so here it is 5v, since no cell voltage is expected to be more than 5v

mha1, 2022-09-12:
Awesome, thanks for the feedback!

offer-shmuely, 2022-09-16:
The pull request has merged
The issue can be closed

closed 2022-09-16 by mha1