EdgeTX/edgetx-sdcard #79: BATT ANALOG Widget not displaying after power cycling radio

LTabora, 2022-09-29:
I've been able to successfully configure the BATT ANALOG Widget on my radio, nevertheless once I power cycle the radio, the widget isn't displayed and I have to configure it again.

Oddly, the .yml file does have the respective entry as follows:

`1:
               widgetName: "BattCheck (A"
               widgetData: 
                  options: 
                     0:
                        type: Source
                        value: 
                           source: tele(9)
                     1:
                        type: Color
                        value: 
                           color: 0xF8FC00
                     2:
                        type: Bool
                        value: 
                           boolValue: 1`

I tried placing the widget on its own page (full screen) and I'm observing the exact same behavior.

I'm running EDGETX 2.7.1 on my RadioMaster TX16S Mark II MAX.

LTabora, 2022-09-29:
I believe this is a duplicate of #36

LTabora, 2022-09-29:
Confirmed. I re-downloaded the SD card contents and also deleted the old widget folder to avoid any future/further confusion.

closed 2022-09-29 by LTabora

pfeerick, 2022-09-29:
Glad it's working for you now! :)