EdgeTX/edgetx-sdcard #114: Boxer missing(?)

woutput, 2023-07-28:
Hi all,

Is it me, or is [Radiomaster Boxer](https://www.radiomasterrc.com/collections/boxer-1/products/boxer-radio-controller-m2) missing here? (and I guess that's why it's missing in the list of radios on https://buddy.edgetx.org/ on the "SD Card content" tab?)
Please let me know if I can help in any way.
BTW Thanks for this great firmware! :)

Kind regards,
woutput

woutput, 2023-07-28:
Should it just be added under
_bw128x64.zip (128x64 pixel black and white screen radios)_
since the [Radiomaster webpage](https://www.radiomasterrc.com/collections/boxer-1/products/boxer-radio-controller-m2) states:
_Display: 128*64 Monochrome LCD display_

?

pfeerick, 2023-07-28:
Support for the boxer was added in v2.8.1, and the last "release" SD card pack is for 2.8.0, hence why it is "missing". If you enable showing pre-releases, you'll be able to use choose "Latest" in the dropdown, and get access to a later release adds reference for it. There is nothing special about that release - as you have discovered, it is a 128x64 B&W screen, so the bw128x64.zip from the releases page suits it. It'll also show up shortly without needing to show a pre-release when the 2.9.0 pack is finally released. 

[image]

https://github.com/EdgeTX/edgetx-sdcard/blob/master/sdcard.json#L27C19-L27C19

woutput, 2023-07-28:
Great, thanks @pfeerick !

closed 2023-07-28 by woutput