EdgeTX/edgetx-sdcard #120: Edgetx Buddy Radiomaster Boxer sdcard

Beachragz, 2023-10-24:
I can complete the Providence 2,9 firmware update, but i cannot get the sdcard to finish updating the providence 2.9 sdcard pack.
It keeps stopping and shutting the radio off during the sounds. i hear a crackle sound on the radio and it goes dark off instantly..?
UNDS/en/SYSTEM/0019.wav - DONE
SOUNDS/en/SYSTEM/0176.wav - DONE
SOUNDS/en/SYSTEM/0018.wav - DONE
SOUNDS/en/SYSTEM/0063.wav - WRITING...
SOUNDS/en/SYSTEM/rxko.wav - WRITING...
SOUNDS/en/SYSTEM/sensorko.wav - WRITING...
SOUNDS/en/SYSTEM/0171.wav - WRITING...

pfeerick, 2023-10-25:
Is the radio battery more than half charged (i.e. it's not switching off because it's flat?)?

If you've got a microSD card reader/adapter, I would try using that instead. You can grab the voice pack from here => https://github.com/EdgeTX/edgetx-sdcard-sounds/releases/tag/v2.9.0 ... just pick the `EN` zip of your preference, and unzip it to the microSD card. Or you could try copy the contents of that zip onto the radio manually with the microSD still in the radio, and try your luck that way.

Beachragz, 2023-10-25:
Excuse my phones camera, it loaded further than ever but seems now stuck on writing scripts.. and it started with a red line not green.  Bad SD card..?Sent from Apple iPhone

Beachragz, 2023-10-27:
> Attachments sent via email don't come through on github.
> It sounds like the SD card got corrupted. Do you have a backup of your model/radio settings? If not, copy the MODELS and RADIO folder off if you still can. Then, format the SD card as a FAT32 file system, and try again. It would be better to use a card reader if you can, as copying lots of files on the radio itself can sometimes play up.

Of course i do bot have a backup like i know i should have, also when i download the files from radiomaster, it has  3 items, factory restore modle and settings, sd card content, and a zip file which has all the languages. I only need the en language.  Do i copy only the sd card content over..?Thank you for responding..Sent from Apple iPhone

Beachragz, 2023-10-27:
Also, is the sd card slot reader on the side of the laptop the same thing as an external reader..?Sent from Apple iPhone

pfeerick, 2023-11-09:
> Also, is the sd card slot reader on the side of the laptop the same thing as an external reader..?

Sorry, I missed seeing that question. Yes, if you have an adapter so you can put the microSD into the SD card reader slot of a laptop that should be fine. And yes, you would only want the SD card content form that file from radiomaster. They simply ensure you have everything - radio firmware, MPM firmware and the SD card content - that they provide in one file.

closed 2024-02-07 by pfeerick