EdgeTX/edgetx-sdcard #128: Multicopter Wizard Default Mixes Switched

robotgoat, 2024-02-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When creating a new multicopter/drone model via the wizard tool, the default channels in the Mixes page of the model are switched. The two channels in question are 1 and 4 for aileron (roll) and rudder (yaw). When using a vanilla installation of EdgeTX with the AETR channel mapping, the input tab for a new multicopter model is correct: Ch1: aileron; Ch2: elevator; Ch3: throttle; Ch4: rudder. However, the mixes for aileron and rudder are switched. 

### Expected Behavior

In the mixes tab, the aileron mix should be taking input from the aileron input and the rudder mix should be taking input from the rudder input. 

### Steps To Reproduce

On a TX16S Mk2 transmitter, press model button. Then press new model. The wizard page will appear, press create multicopter. Follow through the default selection to the end. Then press the model button to access the newly created model's settings. Use the page forward button to access the mixes tab and inspect the aileron and rudder mixes' inputs. They will be swapped when compared to the input tab's mapping. 

### Version

2.9.3

### Transmitter

RadioMaster TX16S / TX16SMK2

pfeerick, 2024-02-06:
I can confirm this behaviour using the latest version of the model wizard, with both AETR and TAER default channel ordering... the inputs are in the correct order, but the mixer channels have both Roll and Yaw sources swapped (but the mixes them selves in the correct order).

pfeerick, 2024-02-06:
You can either download the latest sdcard pack soon (once I've seen why the github build is on the fritz) or simply replace the wizard lua with the latest copy from https://github.com/EdgeTX/edgetx-sdcard/blob/master/sdcard/c480x272/TEMPLATES/1.Wizard/5.Multirotor.lua ... goes in the `/TEMPLATES/1.Wizard` folder

commit 6a6358bd (2024-02-06): fix(wizard): Multicoptor AIL and RUD mix sources swapped

Fixes #128
closed 2024-02-06 by pfeerick