EdgeTX/edgetx-sdcard #129: Wizard creates a Logical Switch that triggers a Play Value of zero that won't stop

daveEccleston, 2024-02-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

I create a plane using all defaults, it adds a mysterious logical switch and Special Function Play Value
After binding, the radio says "Zero" every couple of seconds

### Expected Behavior

I expect it not to add mysterious logical switches and Play Value commands, or I should be given an option to create it or not

### Steps To Reproduce

Start the New Aircraft wizard, choose Plane
Leave everything as defaults and create the model
Go to the Logical Switches page, observe there is an L01 switch that was created. 
- V1 is an eyeball with radar lines (?) coming out of it, not sure what that is
- V2 is 365

On Special Functions, a Play Value on the eyeball is created and tied to L01

When I power on and bind my plane, it starts saying "Zero" repeatedly every few seconds
I have to delete the Play Value function to get it to stop

### Version

2.9.3

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

[image]
[image]

pfeerick, 2024-02-07:
This is happening as those parameters are configured in the template, rather than being created by the wizard lua directly. 

@offer-shmuely Do you want to resolve this? The template really shouldn't contain this sort of thing, as it is making more than a few assumptions about the users setup, and be added via the wizard lua, as it can then also elaborate on any setup requirements - i.e. discover and select your voltage sensor after creation, etc.

daveEccleston, 2024-02-07:
@pfeerick All receivers don't have telemetry (this one doesn't) so the wizard is wrong to automatically add logic that plays annoying, repetitive sounds for telemetry that may not exist. I don't get what's funny about that 

I think @offer-shmuely is correct, this setting should be an option in the wizard. And if the user requests it be added, it should trigger a sensor discovery

offer-shmuely, 2024-02-07:
I will try to do it on the wizard.
I am not sure if the api allow it.
My thoughts was that it easier for the novice user to delete the stuff it don't need, that to know how to add it
But I understand now that this approach is annoying some.
Let me see what can be done

daveEccleston, 2024-02-07:
Thanks for the info regarding the template. Once you told me that I was able to edit the template and remove it from my own radio. I'm doing a demo for my club trying to persuade them that opentx/edgetx radios are better (so many spektrum people in my club). They all think it's too complicated, so having to undo stuff it just did would get the nay-sayers saying "see it's too hard and complicated"

pfeerick, 2024-02-07:
You misunderstand... by definition a wizard will do stuff you either don't want to do, or don't understand... hence it called being a  *wizard* ;)

Not all receivers having telemetry, as well as different voltage sensor names was the reason I suggested against this when it was originally done. 

For a novice, you don't want them to have to remove stuff (after figuring out *why* it's doing what it does)... They have enough information overload as it is... you want it to "just work".

offer-shmuely, 2024-02-08:
see PR #131

daveEccleston, 2024-02-09:
I downloaded your changed files and installed them on my radio, works perfectly, thank you!

PR #131 (2024-02-14): plane wizard fix is-arm switch
plane template: fix is_arm switch, 
plane template: remove default LS & CF

Resolves #129, Fixes #130
closed 2024-02-14 by pfeerick