EdgeTX/edgetx-sdcard #130: Plane Wizard disregards Safety Switch setting

daveEccleston, 2024-02-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

When going through the Plane wizard, I select "NO" for the safety switch setting

When the model is created it makes a safety switch regardless

### Expected Behavior

I expect if I choose "No" for safety switch, that none be created

### Steps To Reproduce

Start the wizard
Choose Plane
On the Motor Settings page, change Safety Switch to "No"
Accept all defaults for all other choices and create the model
Go to the Special Functions page, observe that an override switch was created

### Version

2.9.3

### Transmitter

RadioMaster TX16S / TX16SMK2

pfeerick, 2024-02-07:
This relates to #129, in that configuration parameters are in the template, rather than being created as needed by the wizard lua.

offer-shmuely, 2024-02-07:
I will check

pfeerick, 2024-02-07:
Much appreciated :)

offer-shmuely, 2024-02-08:
see PR #131

offer-shmuely, 2024-02-09:
I can see the same bug on the glider as well

PR #131 (2024-02-14): plane wizard fix is-arm switch
plane template: fix is_arm switch, 
plane template: remove default LS & CF

Resolves #129, Fixes #130
closed 2024-02-14 by pfeerick