EdgeTX/edgetx-sdcard #137: issue for soaretx menu

Tonyevoleurt, 2024-03-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On switching on my radio and clicking on second, left hand button, below SYS, which opens settings/menu for Outputs, Wing Alignment, Mixes and batt, Switches, Centre ailerons, and Graph. Instead of the above each box of each widget says " ERROR in background ○: ?:0: attempt to perform arithmetic on a nil value". When no widget on screen it shows 9 of these messages.
Sometimes on switching on some of the boxes are displayed but cannot be opened, it's also random as to which box appears.
Jesper Frickman gave me a partial fix which returned the page to normal but still could not open the widgets 
 Hope someone can help?

### Expected Behavior

To work as before the update

### Steps To Reproduce

See error

### Version

2.9.4

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

[image]
[image]
[image]

pfeerick, 2024-03-10:
What version of the SD card pack are you using? And this is definitely 2.9.4, not 2.10.0-rc1 or a nightly firmware?

Tonyevoleurt, 2024-03-11:
Hi, sorry for late reply, your message came in during the night. I'm in UK
 I can confirm that the firmware is v1.3.3.31 and  edge-tx  vers v2.9.4 "
Providence. Regards Tony

Tonyevoleurt, 2024-03-11:
Further to you question I'm not sure about "sd card pack" but these are the
headings of what I updated.
Edge-tx firmware v2.9.4
Zip file c480x267
Edge-tx - cpn-win64-v2.9.4
Edge-tx-sdcard-sounds2.9.0
Egdetx themes

Just to mention, I put this on RC forums and it seems a few folks are
having the same issue.

Tonyevoleurt, 2024-03-11:
> I believe that the issue with the Lua scripts crashing was fixed and
> merged with #118 <https://github.com/EdgeTX/edgetx-sdcard/pull/118>. I
> don't know why it hasn't made it into the release.
>
> When I used fixed scripts in Companion 2.9.4, another issue appeared where
> the Lua widgets apparently froze/did not respond to fullscreen mode.
>
> You may find his report and the discuusion on RCGroups here:
> https://www.rcgroups.com/forums/showthread.php?4515145-2-9-4-Edge-tx-update%21-No-soarETX-menu%21

Thanks, I'll check it out! Cheers

Tonyevoleurt, 2024-03-11:
Ok. I only put this on last week so are you saying something has been changed since then?

factoryroll, 2024-03-11:
I can confirm the SoarETX GUI is back to normal for me running EdgeTX 2.9.4 and the pre-release/latest SD card files.

jfrickmann, 2024-03-14:
@Tonyevoleurt is also reporting that he no longer can get fullscreen mode for his widgets. I wonder if this could be related to EdgeTX #4263? I assume it was merged into 2.9.2 and have therefore adviced him to try 2.9.1 to see if that helps.

Tonyevoleurt, 2024-03-14:
Hi, just an update! I went to my back up v2.8.5 that I was using before updating, now that has the same issue, which it did not have before! So I went to v2.9.4 and that has the issue that I originally had, locked widgets. Its almost as if its a bug in my radio, if thats even possible and can not be removed by updating or going backwards. I put the sd card contents from v2.10.0 and the soar widgets are locked again, so are the F3K tasks and practice tabs!
Video below shows state with v2.8.5.  But with v2,9.4 and v2.20.0 widgets are locked

https://github.com/EdgeTX/edgetx-sdcard/assets/162825372/d64a0074-9431-4dfa-a0e5-c826e4f2fa1a

Tonyevoleurt, 2024-03-14:
Thanks Jesper, I tried v2.9.1 but just the same, widgets locked!

jfrickmann, 2024-03-14:
Have you tested any other widgets to see if it is something with my widget?
————————
Jesper Frickmann

Tonyevoleurt, 2024-03-14:
Yes all other widgets and functions seem to be fine.

jfrickmann, 2024-03-14:
Well what can I say - somebody messed something up. Right now I don’t even have a beater DLG to use for EdgeTX testing and I have a build and another repair I want done before I get to it.

I can’t even tell you what previous version of EdgeTX to install because they all have various bugs.

I will put it on my list but it will take some time before I get there. I will probably rewrite score keeping to have minimal dependence on other stuff and get rid of the configuration stuff. It’s just too much work to clean up when other developers break stuff.
————————
Jesper Frickmann

jfrickmann, 2024-03-14:
I pulled my TX16S out of storage, formatted the SD-card and loaded 2.9.4 and the "Centurion" v2.10.0-RC1 SD Card Pack on it.

And things worked.

Therefore the problem is something in your end. What I can think of right now is:
- Find and delete all *.luac files under the WIDGETS folder on your SD card
- If that doesn't work, take another backup, format your SD card and install what I did above.
- If that doesn't work, maybe it's a hardware failure?

Tonyevoleurt, 2024-03-14:
Ok thanks Jesper, I have done the same already but I will do it again as
you have described. Thanks Tony

jfrickmann, 2024-03-14:
Otherwise you need to get hold of your nephew!
————————
Jesper Frickmann

frankiearzu, 2024-04-07:
Just FYI, the code in the latest sdcard is compatible with 2.8.x and 2.9.x.  Took the change done by Jasper, and fixed to account for difference in behavior of  model.getCurve(). 

Recently for 2.10,  there was also a problem on the "switches" screen. The way to identify the last switch of the radio no longer works in 2.10.   Just added a PR for it.. issue: #142

Tonyevoleurt, 2024-04-08:
Thanks for the info Frankie. I've given up with my problem. Tried wiping my radio, had someone else try, now on 2.10.2. Must have tried about 15 times to get soar etx working on my radio, it's as if the bug is now in my radio, I know thats not possible but it's what it seems like. I understand its fixed for everyone else! Anyway I've redone my dlg programming myself so I learnt a lot about edge tx in the process. Thanks for all your help.

pfeerick, 2024-05-11:
Hopefully you managed to get to the bottom of this... closing as this _should_ have been resolved via updating the SoarETX files on your SD card.

closed 2024-05-11 by pfeerick

Tonyevoleurt, 2024-05-11:
No, Peter. I have given up on soar ETX. I've changed SD card twice, I've
gone back to previous versions, I've had other guys, who know more than me
try to change versions all to no result. Everything else is great it's just
soar ETX! Tried everything that Jesper suggested so now just not using it!
On the bright side I can now program my dlg's myself! Thanks for all your
help.