EdgeTX/edgetx-sdcard #139: zorro model wizard LUA bug in 2.9.4 and 2.10.0 RC2

franz4949, 2024-03-31:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The last step before saving the new setup "Ready to go?" ist presented on top vinpage, while on bottom of same page is presented "[Enter Long] to confirm". There is no other option than pressing long ENTER. Then wizard starts from beginning. Thus no model can be saved, wizard is running in endless mode.
Whether other b/w radios also are affected by this LUA failure should be checked.

### Expected Behavior

It should be possible to act on the presented question "Ready to go" and dependent from yes/no it should either restart the wizard or switch to the saving option and save the new model by Long Pressing ENTER.

### Steps To Reproduce

start and run model wizard

### Version

2.9.4

### Transmitter

RadioMaster Zorro

pfeerick, 2024-04-01:
> Then wizard starts from beginning. Thus no model can be saved, wizard is running in endless mode.

If by "starts from the beginning", you mean it shows the "Plane, Delta, Multi" prompt again, no it's not running in endless mode, but instead has been restarted due to how scripts are run in the tools menu... they are run when the ENTER button is released, not pressed... i.e. keep the ROLLER/ENTER button press down longer, and you'll see the tools list... and when you *RELEASE* the button the wizard will launch again... just long press RTN to cancel it.

franz4949, 2024-04-01:
When I press ENTER button and hold it down, the TOOLS page shows up with "Wizard Loader" marked, then when releasing the pressed ENTER button the page with select option "Plane Delta Multi Heli" shows up and I must select one of it and so on. No way to save the new model setup at the end of the programming.

franz4949, 2024-04-01:
And as I already wrote I have no chance to answer the question "Ready to go? before pressing long ENTER.

pfeerick, 2024-04-01:
> When I press ENTER button and hold it down, the TOOLS page shows up with "Wizard Loader" marked,

This means the wizard has completed, and has been launched again. When you long press, the settings are saved, the wizard is exited... and IS STARTED AGAIN. Short press RTN to exit, and check your model settings. Please note, when using the wizard loader, it is making changes to the active model... no new model is ever created.

franz4949, 2024-04-01:
Ahh, no new model is created just the active one is saved with its changes (if any). Yes I can confirm that this is done, so I was wrong with expecting a new model being created (maybe in the past it was??). Anyway thanks for immediate responses and clarification and many thanks to all ETx devs for their excellent work, my highest respect and all the best for the future.

closed 2024-04-01 by franz4949

pfeerick, 2024-04-01:
No problem, I'm glad that cleared it up for you. When OpenTX used to show that as part of the model creation, it would have given the appearance that it created a new model, but in actual fact all that was happening is it would be run after creating a new blank model. I'm thinking of re-instating that "run as part of creating a new model" on B&W radios behaviour, but behind an option flag (as the reason it was originally removed was because most users didn't want it, and found it annoying).