EdgeTX/edgetx-sdcard #145: LiHV support for BattAnalog and BattCheck

KingHippoSE, 2024-05-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

BattAnalog and BattCheck will not show 100% at full charge if you use a LiHV battery. Tested with a 6s LiHV, it probably thinks it's a 7s at low charge.

### Describe the solution you'd like

An option in the widget to make it calculate 100% at 4.35V instead av 4.2V.

pfeerick, 2024-05-02:
BattAnalog was updated about a month ago to add a Lithium_HV option. BattCheck already has the maximum number of 5 options so isn't easily possible for that widget yet. You'll get the updated BattAnalog widget if you update your SD pack to the [latest version](https://github.com/EdgeTX/edgetx-sdcard/releases/tag/latest), or at least replace the `WIDGETS/BattCheck` folder on your SD card with the latest version.

KingHippoSE, 2024-05-07:
BattAnalog LiHV values are just LiPo shifted by 0.15V. Here is a suggested change, I merged the LiHV and LiPO curves so they are the same below 40%. LiPO and LiHV have the same curve except the max voltage, this way you can land at say 30% for both battery types and be at the same voltage. Transform used in python:
hv_data = [
    (voltage + (0.15 * (percent / 100) ** 3) if percent >= 40 else voltage, percent)
    for voltage, percent in lipo_data
]
lua to import:
local percent_list_hv = {
    {3.000, 0},  {3.093, 1},  {3.196, 2},  {3.301, 3},  {3.401, 4},  {3.477, 5},  {3.544, 6},  {3.601, 7},  {3.637, 8},  {3.664, 9},  {3.679, 10},  
    {3.683, 11},  {3.689, 12},  {3.692, 13},  {3.705, 14},  {3.710, 15},  {3.713, 16},  {3.715, 17},  {3.720, 18},  {3.731, 19},  {3.735, 20},  
    {3.744, 21},  {3.753, 22},  {3.756, 23},  {3.758, 24},  {3.762, 25},  {3.767, 26},  {3.774, 27},  {3.780, 28},  {3.783, 29},  {3.786, 30},  
    {3.789, 31},  {3.794, 32},  {3.797, 33},  {3.800, 34},  {3.802, 35},  {3.805, 36},  {3.808, 37},  {3.811, 38},  {3.815, 39},  {3.828, 40},  
    {3.832, 41},  {3.836, 42},  {3.841, 43},  {3.846, 44},  {3.850, 45},  {3.855, 46},  {3.859, 47},  {3.864, 48},  {3.868, 49},  {3.873, 50},  
    {3.877, 51},  {3.881, 52},  {3.885, 53},  {3.890, 54},  {3.895, 55},  {3.900, 56},  {3.907, 57},  {3.917, 58},  {3.924, 59},  {3.929, 60},  
    {3.936, 61},  {3.942, 62},  {3.949, 63},  {3.957, 64},  {3.964, 65},  {3.971, 66},  {3.984, 67},  {3.990, 68},  {3.998, 69},  {4.006, 70},  
    {4.015, 71},  {4.024, 72},  {4.032, 73},  {4.042, 74},  {4.050, 75},  {4.060, 76},  {4.069, 77},  {4.078, 78},  {4.088, 79},  {4.098, 80},  
    {4.109, 81},  {4.119, 82},  {4.130, 83},  {4.141, 84},  {4.154, 85},  {4.169, 86},  {4.184, 87},  {4.197, 88},  {4.211, 89},  {4.220, 90},  
    {4.229, 91},  {4.237, 92},  {4.246, 93},  {4.254, 94},  {4.264, 95},  {4.278, 96},  {4.302, 97},  {4.320, 98},  {4.339, 99},  {4.350, 100},  
}

If you want the more mathematically correct percentage, here is a version with a linear shift from 3.0V to 4.35V:
Transform: hv_data = [(voltage + (percent / 100 * 0.15), percent) for voltage, percent in lipo_data]
Lua:
local percent_list_hv = {
    {3.000, 0},  {3.095, 1},  {3.199, 2},  {3.306, 3},  {3.407, 4},  {3.484, 5},  {3.553, 6},  {3.611, 7},  {3.649, 8},  {3.678, 9},  {3.694, 10},
    {3.700, 11},  {3.707, 12},  {3.712, 13},  {3.726, 14},  {3.732, 15},  {3.737, 16},  {3.740, 17},  {3.747, 18},  {3.760, 19},  {3.765, 20},
    {3.776, 21},  {3.786, 22},  {3.790, 23},  {3.794, 24},  {3.800, 25},  {3.806, 26},  {3.815, 27},  {3.822, 28},  {3.826, 29},  {3.831, 30},
    {3.836, 31},  {3.842, 32},  {3.847, 33},  {3.851, 34},  {3.855, 35},  {3.859, 36},  {3.863, 37},  {3.868, 38},  {3.873, 39},  {3.878, 40},
    {3.884, 41},  {3.888, 42},  {3.894, 43},  {3.899, 44},  {3.903, 45},  {3.909, 46},  {3.913, 47},  {3.919, 48},  {3.924, 49},  {3.929, 50},
    {3.933, 51},  {3.938, 52},  {3.942, 53},  {3.947, 54},  {3.953, 55},  {3.958, 56},  {3.965, 57},  {3.975, 58},  {3.981, 59},  {3.987, 60},
    {3.994, 61},  {3.999, 62},  {4.005, 63},  {4.014, 64},  {4.021, 65},  {4.027, 66},  {4.040, 67},  {4.045, 68},  {4.053, 69},  {4.060, 70},
    {4.067, 71},  {4.076, 72},  {4.083, 73},  {4.092, 74},  {4.099, 75},  {4.108, 76},  {4.117, 77},  {4.124, 78},  {4.133, 79},  {4.141, 80},
    {4.151, 81},  {4.159, 82},  {4.168, 83},  {4.178, 84},  {4.190, 85},  {4.203, 86},  {4.215, 87},  {4.227, 88},  {4.239, 89},  {4.246, 90},
    {4.252, 91},  {4.258, 92},  {4.264, 93},  {4.270, 94},  {4.277, 95},  {4.289, 96},  {4.311, 97},  {4.326, 98},  {4.341, 99},  {4.350, 100}
}

Should be added to WIDGETS/BattAnalog/main.lua

KingHippoSE, 2024-05-07:
The merged version:
[image]

Linear shift:
[image]

offer-shmuely, 2024-05-07:
I will take a look

offer-shmuely, 2024-05-07:
the changes to the BatAnalog looks ok
the BattCheck does not support HV  lipo
I will PR it next week

offer-shmuely, 2024-05-08:
take it from here for testing (I do not use HV batteries)
https://github.com/offer-shmuely/edgetx-x10-widgets/tree/master/WIDGETS/BattAnalog

KingHippoSE, 2024-05-13:
Tested this weekend, works fine now! Closing the issue. Thanks for the fix!

closed 2024-05-13 by KingHippoSE