EdgeTX/edgetx-sdcard #147: Selection of model crashes radio (Emergency Mode) v2.10.0 RC 4 (and RC2)

MartinHugh, 2024-05-11:
Since moving from 2.9.4 to 2.10 RC4 one of my models (*MINIDRAK") crashes the radio (TX16S) to Emergency Mode when it is selected.
This is reproducible but only on this model. All the others appear to work fine.

This problem model worked fine 1 week ago when the radio was on 2.9.4

The initial discussion on this is at
https://discord.com/channels/839849772864503828/1238475482589302844/1238475482589302844
... the radio, models files and model image are uploaded there.
[EDIT: and now also there is a link to the sdcard there too]

I have also ruled out a problem with sounds files ( some of which would play when the model is selected) by removing all sounds temporarily from the SOUND folder. No change, this model crashes the radio, none of the others do

I have a second (backup) radio which is identical except it is on RC2. This behaves the same as the RC4 one. That is, the radio crashes when this model is selected.

MartinHugh, 2024-05-11:
This is resolved for me now (see discord discussion, thank you all)

I am happy for this to be closed now, but if it is not already there, I think a note in the next release notes should include a mention that the LUA scripts should be the latest ones.

MartinHugh, 2024-05-12:
As this is potential problem is referenced in the 2.10.0 final release, I will close this.

closed 2024-05-12 by MartinHugh

pfeerick, 2024-05-12:
Thanks (and thank you for reporting it also) :)