EdgeTX/edgetx-sdcard #148: bug in wizard create new model multirotor

ror1948, 2024-05-12:
bug in wizard create new model multirotor 
[image]

pfeerick, 2024-05-12:
With what version of firmware, and what version of the sd card pack? Using firmware 2.10, and sdcard pack latest, I don't get any errors... the error sounds like it is due to a mismatch between firmware and version of the wizard, as they got quite a few updates about three months ago.

ror1948, 2024-05-12:
ok it works ok with just released 2.10.0 but why is the yaw ch1 and ail is ch4 ?

ror1948, 2024-05-12:
also how do you run the templates?thanks

pfeerick, 2024-05-12:
> but why is the yaw ch1 and ail is ch4 ?

Because your default channel order in Radio Settings -> Radio Setup is set to R??A. 

> also how do you run the templates

When creating a new model, wizards are the first group. If you save any models as a template, you will have a `2. Personal` entry also, and you can access those saved models from there. 

cf. 
https://manual.edgetx.org/color-radios/select-model#creating-a-new-model
and 
https://manual.edgetx.org/color-radios/select-model#selecting-and-managing-existing-models

ror1948, 2024-05-13:
no it is AETR
yes i know all that and that works but how o you execute the wizard without selecting a new model?

thanks

 
 

https://manual.edgetx.org/color-radios/select-model#creating-a-new-model
https://manual.edgetx.org/color-radios/select-model#selecting-and-managing-existing-models

pfeerick, 2024-05-13:
You are not supposed to. It is only supposed to be run on an unconfigured
blank model, otherwise the settings may  be messed up.

If you insist on doing that, you can run the wizard by browsing to it via
the radio settings sd card browser.

ror1948, 2024-05-13:
yes i have run it by browsing but that is a pain.I always knew it acted on current model and that was the design concept then.I actually would knowing run on a current model when wizard was in tools
how about a launch Lua for wizard in tools
so the design concept has changed to only run on new model?Do not think it is a good idea to protect the user so much. Just warn users when Lua is run

thanks

pfeerick, 2024-05-13:
> but why is the yaw ch1 and ail is ch4 ?
> no it is AETR

With AETR default channel order, Roll is CH1
[image]
 ... and Yaw is CH4...
[image]

> how about a launch Lua for wizard in tools

You can modify the launcher that was present in 2.8 for the new paths if you want that. 
https://github.com/EdgeTX/edgetx-sdcard/blob/v2.8.0/sdcard/c480x272/SCRIPTS/TOOLS/WizardLoader.lua

> so the design concept has changed to only run on new model?Do not think it is a good idea to protect the user so much. 

This was always the design really... was the case in OpenTX, and then initially in EdgeTX. It was removed for a while, and I added the loader so that it was easier to find the wizards again. Once the templates were added, wizards were re-added again as part of the new model workflow. Standalone launcher is unlikely to be re-added as this is not the intended use.

closed 2024-05-13 by pfeerick

ror1948, 2024-05-13:
ok thanks
i am modifying the paths in the wizard loader but that is a big edit.the directory structure is very different between the two.you can not just change the root directory path.old location had directory of ICONS while the new has PLANE,MULTIROTOR etc

AETRyes I knowDo not know why but I had to edit the input screen to fixI have been using an old nightly build for months tx16sN1-b0e64c8so maybe that had something to do with it.now on latest releasethanks for your help

 
 

screenshot_tx16s_24-05-13_15-15-32.png (view on web)
screenshot_tx16s_24-05-13_15-15-42.png (view on web)

https://github.com/EdgeTX/edgetx-sdcard/blob/v2.8.0/sdcard/c480x272/SCRIPTS/TOOLS/WizardLoader.lua

ror1948, 2024-05-13:
no, modifying the paths in the wizard loader will not work.old wizard just had a LUA but templates uses LUA and YMLDo not see a main program in templates for wizardold wizard uses pictures of UAV while templates version just has a list on screen.

 

 
 

https://github.com/EdgeTX/edgetx-sdcard/blob/v2.8.0/sdcard/c480x272/SCRIPTS/TOOLS/WizardLoader.lua

pfeerick, 2024-06-20:
You can't get the benefit of anything that was done in the templates, as that is specific to the templates - not the wizards, but to re-instate the wizard loader as it existed prior to being dropped in 2.9, it is merely a matter of grabbing the old icons, and changing the paths in two places in the loader file to use the new wizards. Note this doesn't give access to the Wing wizard, simply because it is also new in 2.9, so there is no icon for it. The code should be self-explanatory enough that if you have or make a suitable icon, that can be added also. 

[wizard-2.9-layout.zip](https://github.com/user-attachments/files/15909491/wizard-2.9-layout.zip)