EdgeTX/edgetx-sdcard #161: EL18 support more Theme LUA“320*480”

fxk666, 2024-07-25:
Which LUA can support in EL18？

pfeerick, 2024-07-25:
I'm not sure the question... Lua or themes (as the title says themes and lua?) If you want themes... you want to look at https://github.com/EdgeTX/themes

EL18 / 320x480 has always had a much more limited audience, so there are very few Lua scripts that are specifically adapted for that screen orientation / and touch only nature. https://github.com/yaapu/FrskyTelemetryScript and https://github.com/iNavFlight/OpenTX-Telemetry-Widget are the only two big ones I know of... ELRS Lua should work just fine also.

fxk666, 2024-08-07:
I am looking for a way to reproduce this situation. Thank you for your reply

pfeerick, 2024-08-08:
What situation are you trying to reproduce?

closed 2024-10-10 by pfeerick