EdgeTX/edgetx-sdcard #166: BattAnalog widget shows 1% when battery is fully charged

LoooooKe, 2024-10-03:
The BattAnalog widget shows 1% when the battery is fully charged. I narrowed it down to the following line:

https://github.com/EdgeTX/edgetx-sdcard/blob/b236171b6fae5ff247748acb303b8177c61aaba3/sdcard/c480x272/WIDGETS/BattAnalog/main.lua#L257

If the voltage is equal or even .01 above the value in voltageRange, the method returns the wrong cell count, which then leads to a wrong calculation of the cellPercentage value here:

https://github.com/EdgeTX/edgetx-sdcard/blob/b236171b6fae5ff247748acb303b8177c61aaba3/sdcard/c480x272/WIDGETS/BattAnalog/main.lua#L326

My suggestion is to round singleVoltage to one decimal and change the '<' to a '<='. This should fix the issue for all battery types.

offer-shmuely, 2024-11-16:
can you explain why it is happen in your scenario?
what voltage do you get on full, and what type of battery?

your suggestion can work, thanks
an another way will be to just enlarge a little the voltageRange array values.
but lets first understand way it is happening

[image]

LoooooKe, 2024-11-17:
Battery is 2S LI-Ion and the total voltage is like 8.41. I tried to enlarge the range but this didn't work because the cellCount is wrong. That's why I came up with this working solution:
```
-- Only invoke this function once.
local function calcCellCount(wgt, singleVoltage)
    local voltageRanges = voltageRanges_lipo
    local singleVoltageRounded = math.floor(singleVoltage*10)/10

    if wgt.options.Lithium_Ion == 1 and wgt.options.Lithium_HV == 0 then
        voltageRanges = voltageRanges_lion
    end
    if wgt.options.Lithium_Ion == 0 and wgt.options.Lithium_HV == 1 then
        voltageRanges = voltageRanges_hv
    end

    for i = 1, #voltageRanges do
        if singleVoltageRounded <= voltageRanges[i] then
            log("calcCellCount %s --> %s", singleVoltageRounded, i)
            return i
        end
    end

    log("no match found" .. singleVoltage)
    return 1
end
```

pfeerick, 2024-11-24:
No always... some li-ion are 4.1, some are 4.2.

offer-shmuely, 2024-11-25:
So i need to enlarge the voltage  level of separation

offer-shmuely, 2024-11-30:
I updated the rates of the li-ion ranges
please try the test version at
https://github.com/offer-shmuely/edgetx-x10-widgets/tree/master/WIDGETS/BattAnalog

and update me here.

offer-shmuely, 2024-12-14:
@LoooooKe - reminder

offer-shmuely, 2024-12-28:
fixing in PR  #181 
https://github.com/EdgeTX/edgetx-sdcard/pull/181

PR #181 (2025-01-19): fix(BattAnalog): adapt lion voltage to 4.2 lion, graceful error if < 2.11
Fixes #166, fixes #180
closed 2025-01-19 by pfeerick