EdgeTX/edgetx-sdcard #167: Frsky lua script and telemetry data for rb30/35s not working on taranix x9d se plus 2019

jorgeneo560, 2024-10-04:
tx: taranix x9d se 2109
rx: frsky sr10+
Device: frsky rb30 safe flight system board

The lua scrip avaible from frsky site for configuring and calibrating the rb30/35s is not working, when i run the scrit it trows error at literal '>', also when i run discover new sensors it dosent show up any telemetry data of the rb board, while the same setup with the tandem x20 it show all the data, im runing the lastes edge tx firmware up to date.

offer-shmuely, 2024-11-16:
the script for the taranis was not written for the RB30
it was written for the S6R & S8R

jorgeneo560, 2024-11-17:
im speaking about the script that is on frsky site, when you run the script to configure the rb30 devices dosent work, it trow that error, up today, the only way to configure a rb30 power board is with a tandem radio, you cant do anything with edgetx, even firmware update via internal sport, it would be nice to make it work becouse for now you  are forced to buy a tandem radio if you planing to use this devices