EdgeTX/edgetx-sdcard #170: Jumper T15  with  RM ER6GV Rx -Locate Model tool does not work

SteveintheIoW, 2024-11-02:
I have a new T15 tx (tried Edgetx fw 2.10.1 and 2.10.5) working with a RM ER6GV ELRS receiver.  The Setup works - control,  telemetry and Vario all work fine BUT unable to get Locate Model tool to work.  The tool just bleeps every second and shows 0dB.
I've tried clearing out all models, updated SD card pack. Is there some Tx Rx compatibility issue or LUA bug

offer-shmuely, 2024-11-16:
can yo send picture of the sensors screen

SteveintheIoW, 2024-11-18:
Video is here

[https://www.dropbox.com/scl/fo/9g2nt96le6nffk18s6ieq/AG7gAIcyZsLxhG5lZc9vutk?rlkey=fecnvj9x8tc0b0319nffbgesl&st=0a4iwgpp&dl=0](url)

And sensor data screens

[image]
[image]
[image]

These models are using mateksys and RM versions of the same ER6GV receiver.  Both models seem to work fine, Vario and alt data readouts etc seem fine but the locator tool just sits at 0dB as shown .  

Any help appreciated! Cheers Steve

SteveintheIoW, 2024-11-18:
https://www.dropbox.com/scl/fo/9g2nt96le6nffk18s6ieq/AG7gAIcyZsLxhG5lZc9vutk?rlkey=fecnvj9x8tc0b0319nffbgesl&st=0a4iwgpp&dl=0

SteveintheIoW, 2024-11-18:
Video link  again..

https://www.dropbox.com/scl/fi/154qvnpgzq3bdk3r5lcxs/VID_20241118_105521152.mp4?rlkey=l1cep62o7t3a0e9lmhxq6y16m&st=3h1k4hry&dl=0

offer-shmuely, 2024-11-18:
> @offer-shmuely the 480x320 version only looks for RSSI sensor by name.
> 
> Additionnaly, why not using getRSSI() ? I returns a form off RSSI automatically depending on module type.
> 
> https://luadoc.edgetx.org/part_iii_-_opentx_lua_api_reference/general-functions-less-than-greater-than-luadoc-begin-general/getrssi

since I like to show what is used 
since i also like sometime to allow select the signal, i.e. let the user choose ant1 or ant2
since the mix max values are different 0-100 or -40 to -112

SteveintheIoW, 2024-11-18:
Excellent and thank you so much - it looks like the version you linked me to has fixed it on the Jumper T15....I haven't had much time to exhaustively test it but I am now seeing a bar Graph display.  
Many thanks again for your help and the all the great work and effort that you and the team have done on this project. Its incredible.  Cheers Steve

offer-shmuely, 2024-11-18:
fixed in PR (update script also in 480x320)
this bug can be closed when the PR merged
https://github.com/EdgeTX/edgetx-sdcard/pull/174

PR #174 (2024-11-19): model locator: display error info when no signal found
Fixes #170
closed 2024-11-19 by pfeerick