EdgeTX/edgetx-sdcard #171: Lost sound on SD card with 2.10.5 update

JayBu0517, 2024-11-09:
After updating, and having new files copied over from edge TX companion, sounds files did not copy over.  So went and manually moved sound files, and if looking through SD card contents, they are there under sounds, but when trying to attach a a track to a switch in special function, it tells me no sounds found. Cant find a resolve for this.

pfeerick, 2024-11-10:
Check the layout of your SD card. In the top level, you should have MODELS, RADIO, SOUNDS, etc. Inside SOUNDS, you should have a two letter directory , eg "EN". inside that would be a bunch of wav files, and a SYSTEM subdirectory with a bunch more wave files. The error message you are getting is indicating this is not the case.

pfeerick, 2025-01-19:
Any progress / updates on this? Will close issue next week if no new information.

closed 2025-06-22 by pfeerick