EdgeTX/edgetx-sdcard #172: in Wizard for WING: Switch picked for dual rates is not shown in summary screen or programmed in model

ror1948, 2024-11-13:
I have updated the program in the Lua file "3.Wing.lua" attached so this fault does not occur.

Changes below:

-- WAS at line 452
-- updateInputLine(defaultChannel_0_AIL, 0, expoVal, 100, "SC" .. CHAR_UP)
-- updateInputLine(defaultChannel_0_AIL, 1, expoVal, 75 , "SC-")
-- updateInputLine(defaultChannel_0_AIL, 2, expoVal, 50 , "SC" .. CHAR_DOWN)
-- updateInputLine(defaultChannel_0_ELE, 0, expoVal, 100, "SC" .. CHAR_UP)
-- updateInputLine(defaultChannel_0_ELE, 1, expoVal, 75 , "SC-")
-- updateInputLine(defaultChannel_0_ELE, 2, expoVal, 50 , "SC" .. CHAR_DOWN)

-- IS at line 452
updateInputLine(defaultChannel_0_AIL, 0, expoVal, 100 , ElevronFields.dr_switch.avail_values[1 + ElevronFields.dr_switch.value]..CHAR_UP)
updateInputLine(defaultChannel_0_AIL, 1, expoVal, 75 , ElevronFields.dr_switch.avail_values[1 + ElevronFields.dr_switch.value].."-")
updateInputLine(defaultChannel_0_AIL, 2, expoVal, 50 , ElevronFields.dr_switch.avail_values[1 + ElevronFields.dr_switch.value]..CHAR_DOWN)
updateInputLine(defaultChannel_0_ELE, 0, expoVal, 100 , ElevronFields.dr_switch.avail_values[1 + ElevronFields.dr_switch.value]..CHAR_UP)
updateInputLine(defaultChannel_0_ELE, 1, expoVal, 75 , ElevronFields.dr_switch.avail_values[1 + ElevronFields.dr_switch.value].."-")
updateInputLine(defaultChannel_0_ELE, 2, expoVal, 50 , ElevronFields.dr_switch.avail_values[1 + ElevronFields.dr_switch.value]..CHAR_DOWN)

-- added these lines
 at line 385
--dual rate sw
if ElevronFields.is_dual_rate.value == 1 then 
drawNextLine("Dual Rate SW", nil, ElevronFields.dr_switch.avail_values[1 + ElevronFields.dr_switch.value])
end

[updated_3.Wing.zip](https://github.com/user-attachments/files/17725942/updated_3.Wing.zip)

ror1948, 2024-11-13:
moved to
https://github.com/pascallanger/DIY-Multiprotocol-TX-Module/issues/1043

pfeerick, 2024-11-19:
Sorry for the delay in responding to this... this is the correct repo to do this, rather than edgetx/edgetx or pascallanger/DIY-Multiprotocol-TX-Module

offer-shmuely, 2024-11-30:
PR #178

PR #178 (2024-12-15): wing wizard: fix dual rate switch display & setting
fix dual rate switch display on config summary
and correct actual setting 

[image]

Fixes #172
closed 2024-12-15 by pfeerick