EdgeTX/edgetx-sdcard #182: Wizard Script Error Flysky EL18

Aeroengineer1, 2025-01-05:
**TLDR Version**
1. In the glider and plane wizards for the 320x480 SD Card, the Lua scripts need to replace DEFAULT_COLOR with COLOR_THEME_PRIMARY1
2. The background png needs to be properly sized from 480x270 to 320x480.
3. Screen orientation will likely cause important graphics to be off the screen
4. EL18 uses the trim stick to have page right or page left.  This does not seem to advance the screens in this script

### **Basic Details**
**Radio:** Flysky EL18
**EdgeTX Version:** 2.10.5 (stated as release chan in EdgeTX Companion 2.10.5)
**Update Method:** Updated via downloading SD Card contents via EdgeTX Companion, then initiating bootloader mode at the radio

**Issue:** 
When trying to use the wizard for model creation, it does not show up in the list of tools as shown in the [manual](https://manual.edgetx.org/color-radios/radio-settings/tools).  Neither do the other two tools.  I can, though, find the script by going into the sd card to SCRIPTS/Wizard/plane/wizard.lua or wizard.luac.

When I first attempted to do this, the .luac file was not in the folder.  I checked to see if somehow I missed this file when I was copying the files from EdgeTX Companion to the SD card, but I had not missed the file.  The .luac file appears after the first attempt to run the file.  When I select the file and tell it to execute, I see the screen of the wizard open, but then an error message appears.  They are slightly different between the .lua vs .luac files.  I also get these error messages when trying to run the glider wizard script.

**Initial .lua Error Message**
Script syntax error
Wizard/plane/wizard.lua:74:
attempt to perform arithmetic 
on field 'DEFAULT_COLOR' (a 
nil value)

**.luac Error Message**
Script syntax error
?:0: attempt to perform 
arithmetic on field
'DEFAULT_COLOR' (a nil value)

I am not that great of a programmer, nor do I have experience in Lua, but from what I can see, this is in the function redrawFieldsPage.  Looking at the version of this script that is for the PL18 (different screen orientation) this variable is called COLOR_THEME_PRIMARY1.  This looked like a global type variable that could be used.  I found the COLOR_THEME_PRIMARY1 in the [EdgeTX Color Constants Lua Reference Guide](https://luadoc.edgetx.org/part_iii_-_opentx_lua_api_reference/constants/color-constants).  There were deprecated constants listed there as well, but I did not find DEFAULT_COLOR as a constant.  I then went over to OpenTX to see if I could find that listed as a constant.  I could not.

I then tried to just take the lua file from the 480x320 version of the SD Card, and it worked right off.  The only "issue" is that the background .png is the wrong size for the 320x480 screen.  When I looked at the background image asset in the 320x480 SD Card download it measures 480x270.

This issue of screen layout also causes the location of the different graphics to likely be off the screen in a vertical layout method.

This also leads to the last issue.  The right trim stick has the selection button as well as page right and page left.  The selection button and moving up and down in the options works with the stick, but going to the next page by hitting right or left does not seem to work.

**TLDR Version**
1. In the glider and plane wizards for the 320x480 SD Card, the Lua scripts need to replace DEFAULT_COLOR with COLOR_THEME_PRIMARY1
2. The background png needs to be properly sized from 480x270 to 320x480.
3. Screen orientation will likely cause important graphics to be off the screen
4. EL18 uses the trim stick to have page right or page left.  This does not seem to advance the screens in this script**

choekstr, 2025-04-19:
Can confirm this is still an issue in 2.10.6 with the EL18. I was able to get rid of the error but as you said, can't advance the wizard.  I didn't resize any images as the first screen can't be advanced so I didn't bother. 

The wizard also doesn't show up under the new model choices like in the tx16s. Instead I had to go into the SD card widgets and execute the lua script manually. 

Would be nice to get this working in a future version or a commit we can replicate ourselves.

pfeerick, 2025-07-08:
This should all pretty much be resolved by #198... only one issue outstanding still regarding the multirotor wizard graphics.

closed 2025-07-08 by pfeerick