EdgeTX/edgetx-sdcard #186: Images Not Seen On SD Card

Southportpat, 2025-02-13:
Hi

Have created some imaged on the SD card in the Images folder - all images are less than 7 character long and 192 pixels or less in size. - have about 25 images on there.

My transmitters a Radiomaster TX16S can see some of the images but not all - but they do exist on the SD cards.

Have tried both jpg and png formats

Anyone any advise please

pfeerick, 2025-02-14:
Just FYI, this is would have been more appropriate at https://github.com/EdgeTX/edgetx like with your other post, as this won't be related to the SD card content. 

So when you say you 

>  can see some of the images but not all

where/when? Do you mean when trying to pick one to use it as an model image? What about when you go to to Radio Settings -> SD Card (second tab) -> IMAGES? Can you see them listed there? And do the image previews show?

offer-shmuely, 2025-04-25:
ass peeter said, you should create a new issue on  https://github.com/EdgeTX/edgetx
and close this one.
the logic: this error (if exist) can not be fix by changing a lua script (this repo)
I also suggest to supply more info, the zip of the files, picture of them in the folder, picture of where you expect it to be exist...

offer-shmuely, 2025-04-26:
@pfeerick I think you should close it here. 
since it can not be fixed in this repo anyway.
if it is really a bug, it can be fixed only on the other repo as you wrote