EdgeTX/edgetx-sdcard #203: GX-12 no rgb leds working with 2.11

kakashi2018, 2025-05-04:
Radiomaster gx-12 radio updated to latest 2.11.0 release with sd card contents and see that rgbled lua are no longer recognized in the radio. they were working with factory firmware when set under global functions, now it is not even showing rgb leds option

3djc, 2025-05-04:
Ideally they should be removed for gx12 sd content, but I think are there even for radio without rgbleds (unsure)

pfeerick, 2025-05-04:
> Ideally they should be removed for gx12 sd content, but I think are there even for radio without rgbleds (unsure)

They are present for all c480x320 and bw128x64 targets. Since they are only accessed by the RGB SF, and only take a small amount of space, it is really not necessary to complicate the situation by making it so there are even more sd card pack variations. 

Since the underlying issue is really in the firmware, not the sd card pack, and is being tracked here edgetx/edgetx#6197, so closing this issue.

closed 2025-05-04 by pfeerick