EdgeTX/edgetx-sdcard #208: Edge TX

woodman352, 2025-05-14:
I wrote the following to Radiomaster Support Team.

Since converting to EdgeTX my Gyro and Non Gyro receiver models transferred OK and work OK.  

However, trying to set up a new Gyro receiver (FrSky ArcherPlus SR8) has been a problem.  The SYS / TOOLS / FrSkyS8R/S6R/Calibrate. box freezes on the first page whether a receiver is connected or not.  I can't see a FrSky SR8 box. 

I have to turn the Tx off and back on to Unfreeze.

I have updated from Centurion to Jolly Mon.  I have updated the internal 4 in 1 module.  I have updated Lua scripts.

I'm not sure whether I've caused the problem during the initial upgrade from OpenTx to EdgeTx Centurion.

Does anyone recognize the problem.

Is it advisable to get a new SD Card and start afresh, Although I'm not too sure how to remove the radio firmware.

I would be grateful for your advice

Kind Regards
Roger Saunders**_

Hi Roger Saunders,

Sorry, we are a hardware manufacturer, and the software system is developed and provided by the open-source organization EdgeTX. I checked the SD card content of EdgeTX and saw that currently only the setting script for Frsky S6R/S8R is available, but not the setting script for Frsky SR8. Since Frsky SR8 is too new, perhaps EdgeTX has not yet provided the lua script for setting SR8. If you have any questions regarding this aspect, you can join the EdgeTX Discord discussion group and ask the developers.

This is the content of the latest EdgeTX SD card I queried. It can be seen that there are no scripts related to SR8, only setting scripts for S6R and S8R can be seen.

https://github.com/EdgeTX/edgetx-sdcard/tree/master/sdcard/c480x272/SCRIPTS/TOOLS

Kind regards,

Radiomaster support team

I have checked the contents of the Script/Tools file as advised by Radiomaster.  Looked at the files shown by your pfeerick d76e5a8 and all seems to be in place - not checked the code.   I have also seen a page indicating a "patch" if that's the right word from your FrankieArzu but am reluctant to try it at the moment in case my problem isn't the problem Frankiearzu is trying to solve.  The screen freeze occurs whether a Rx is connected or not.

I would appreciate your advice

Kind Regards
Roger Saunders

pfeerick, 2025-05-17:
> I have to turn the Tx off and back on to Unfreeze.

Did you try holding RTN? Most likely the transmitter is not frozen, but merely the Lua script is, and a long press (more than a second) of RTN will force exit any/all standalone Lua scripts. 

> FrSky ArcherPlus SR8

So, to be clear, you are talking about this receiver? https://www.frsky-rc.com/product/archer-plus-sr8/

I would strongly suggest you either update to the latest sd card pack marked "Latest" (not the 2.11.0 one) and try the newly added (in the last few days) FrSky Gyro suite for the SR10+/SR12+, which may also work for your SR8. 

Also please note, we don't work for RadioMaster, and I am mostly only the maintainer for this code repository, Frankie and Offer are responsible for most of the FrSky Gyro suite code.

woodman352, 2025-05-18:
Thank you for your reply.

I have joined the Edge Discord group and Frankie Arzu has also replied with some download advice and suggestions.  I have not been able to download his latest updates as yet.