EdgeTX/edgetx-sdcard #217: Flights Widget

Taes59, 2025-06-25:
In the latest 2.11 update, the Flights widget has disappeared from the screen. The SD card widget still appears.

Taes59, 2025-06-25:
I'm trying to reactivate the flights widget but it doesn't appear in the folder.

offer-shmuely, 2025-07-01:
What version of etx do you run?

offer-shmuely, 2025-07-01:
Did not understand

Taes59, 2025-07-01:
2.11.2

offer-shmuely, 2025-07-02:
Strange
Can you send the zip of your widget folder?

offer-shmuely, 2025-07-03:
you are using an old widget versions that was not designed to edgeTx 2.11
* Flight 1.1 instead of 1.7
* BattAnalog 0.7 instead of 1.6
* BattCheck 0.6 instead of 0.9
* ModelName 0.3 instead of 0.8
* Value2 0.8  instead of 0.12

please update all and try again

Taes59, 2025-07-03:
OK, en la ultima actualizacion 2.11, en el contenido de la tarjeta SD, ya estan actualizados?
Gracias

offer-shmuely, 2025-07-03:
All of the once on the git scdcard is working correctly on 2.11
The one that you use, do not.
You need to update