EdgeTX/edgetx-sdcard #225: crossfire.lua missing from SD card release v2.11.0

Gelassen, 2025-07-16:
### Issue Summary

The latest SD card release for EdgeTX `v2.11.0` does **not include the `crossfire.lua` script** in the `SCRIPTS/TOOLS` folder. This script is essential for configuring TBS Crossfire modules from within the TX16S running EdgeTX.

### Steps to Reproduce

1. Go to https://github.com/EdgeTX/edgetx-sdcard/releases/tag/v2.11.0
2. Download `c480x272.zip` (or any screen-specific zip)
3. Unzip and inspect `SCRIPTS/TOOLS` folder
4. Observe that `crossfire.lua` is missing

### Expected Behavior

The SD card pack should include `SCRIPTS/TOOLS/crossfire.lua`, which is used to access the Crossfire TX menu and bind receivers.

### Actual Behavior

The file is missing, making it impossible to access Crossfire settings via the TX UI unless manually added from other sources (e.g. TBS Agent X).

### Request

Please restore `crossfire.lua` to the official SD card pack or provide guidance on where to obtain the latest compatible version for EdgeTX `v2.11.0`.

### System Info

- Radio: RadioMaster TX16S
- Firmware: EdgeTX 2.11.0
- Screen layout: `c480x272.zip` used
- Crossfire TX module installed and working

Thanks for your support!

pfeerick, 2025-07-17:
This has not been present since at least release v2.5.0, i.e. the last four years. For configuring TBS Crossfire/Tracer, you should be using TBS Agent Lite, as provided by TBS here => https://www.team-blacksheep.com/products/prod:agentx

The ExpressLRS Lua can also be used to configure Crossfire (and vice versa).

Gelassen, 2025-07-17:
Thank you for clarifying it.