EdgeTX/edgetx-sdcard #230: Helicopter wizard gives wrong mixer setup for T Gain

ror1948, 2025-08-31:
select NEW in Manage Models and then select wizard.
then select helicopter.
Any helicopter type will produce an incorrect mixer setup for T Gain.
For the source of T Gain it shows "smB"

[screen-2025-08-30-235644.bmp](https://github.com/user-attachments/files/22062416/screen-2025-08-30-235644.bmp)
[screen-2025-08-30-235657.bmp](https://github.com/user-attachments/files/22062415/screen-2025-08-30-235657.bmp)

jrwieland, 2025-09-02:
Heli wizard sources updated to the correct source for each SD card type 
See [PR # 231](https://github.com/EdgeTX/edgetx-sdcard/pull/231)
Thank you for reporting it.

jrwieland, 2025-11-28:
Peter this issue was resolved with  pr #231 , this can be closed